zmkfirmware/zmk-studio #35: Standalone desktop application or website?

and-elf, 2024-09-29:
Is this supposed to be a standalone desktop application, that users install, or a website - like VIA?
I think it could be both. A desktop application for developers and a website for most (?) users. 

The zmk GitHub hooks could probably be used in the background, to build the binaries, and then supplied to the gui when the build is complete.

petejohanson, 2024-09-29:
@and-elf We already have an app version of this available, we just haven't formalized a first "release" since this is still in alpha testing. You can grab the latest from the latest build in https://github.com/zmkfirmware/zmk-studio/actions

and-elf, 2024-09-29:
Oh, cool!
So, I supply a patch for the layout of my keyboard (ergodash), and then I should be able to use the webapp?

petejohanson, 2024-09-29:
> Oh, cool!
> So, I supply a patch for the layout of my keyboard (ergodash), and then I should be able to use the webapp?

Yes. But please note this is *alpha* testing stage, and using it accepts that there may be bugs still.

and-elf, 2024-09-29:
Well, obviously. I will try and see how it works. Anyway, very good job! It will absolutely make zmk a lot more convenient, and be hopefully adopted by many OEMs currently using via.

closed 2024-09-29 by and-elf