zmkfirmware/zmk-studio #40: drop down auto closing

minhe7735, 2024-10-01:
Issue: Drop down auto closing when clicking the arrow button

System: Desktop app (Windows 11)

Details: When cursor is placed at location where the start of the string to the location of the cursor does not fit inside the text box, the dropdown menu auto closes.

Video: https://github.com/user-attachments/assets/d0cc3716-52b3-48e9-9f30-f1593d777ea1
  -video shows that if the cursor is placed after "w", which falls outside the textbox size it starts auto closing but before "w" is fine

petejohanson, 2024-10-07:
@minhe7735 Can you please test the deploy preview https://deploy-preview-52.preview.zmk.studio/ from #52 and confirm this fixes things for you? I tested here and was able to reproduce in Edge on W11, and then confirm that the above fix resolved the issue.

Thanks!

minhe7735, 2024-10-07:
Fixed: tested on browser and windows desktop app

closed 2024-10-07 by minhe7735