zmkfirmware/zmk-studio #74: compiler error with: mods = <(MOD_RCMD|MOD_LCMD)>; but no error with mods = <(MOD_RGUI|MOD_LGUI)>;

ladcash, 2024-11-04:
**Describe the bug**
1. mod-morph definitions using RCMD & LCMD keycodes: ...mods = <(MOD_RCMD|MOD_LCMD)>; fail to compile with error message: "devicetree error: /tmp/zmk-config/config/corne.keymap:49 (column 185): parse error: expected number or parenthesized expression.", where the line number is correct, the column number has no relationship to the error source, & the only clue as to the source is in the kconfig device tree file: "dtu: d_to_up { compatible = "zmk,behavior-mod-morph"; #binding-cells = <0>; bindings = <&kp ((((0x07) << 16) | (0x07)))>, <&kp (((((0x07) << 16) | (0x52))))>; mods = <(MOD_LCMD|MOD_RCMD)>; };."
2. the Keymap Upgrader identifies no problems with LCMD/RCMD
3. the identical code lines using RGUI & LGUI keycodes: ...mods = <(MOD_RGUI|MOD_LGUI)>; compiles perfectly.

**To Reproduce**
Steps to reproduce the behavior:
1. Compile:
h2l: h_to_left { compatible = "zmk,behavior-mod-morph"; #binding-cells = <0>; bindings = <&kp H>, <&kp LEFT>; mods = <(MOD_RCMD|MOD_LCMD)>; };
3. See error

**Expected behavior**
obvious

**Screenshots**
not applicable

**Environment (please complete the following information):**
 - M1 Max MacBook Pro Sequoia 15.x.x...
 - Safari 18.x.x...
 - GitHub
 - Corne with nice!nano v2

**Additional context**

caksoylar, 2024-11-04:
I think you opened this in the wrong repo, maybe yolu meant to open in https://github.com/zmkfirmware/zmk ?

caksoylar, 2024-11-04:
FWIW, the mod flags are not necessarily the same as modifier keycodes. The list is documented at https://zmk.dev/docs/keymaps/behaviors/mod-morph#mods and it doesn't include `MOD_*CMD` as valid flags.

ladcash, 2024-11-04:
Thanks!! Cem,

 

It’s worth a lot – the right answer always is. I sped past this mods list in the mod-morph section & assumed that the same keycode is sent with platform interpretation differences. This seems to be the case as all of my mod-morphs with GUIs work perfectly.

 

Be Excellent, Michael

closed 2024-11-04 by ladcash