zmkfirmware/zmk-studio #83: Linux: amd64 AppImage fails to launch

krikun98, 2024-11-12:
**Describe the bug**
amd64 AppImage fails to launch: I can see a black window for a fraction of a second, and then it just dies.
From the console the following log is produced:
```
./zmk-studio_0.1.0_amd64.AppImage                                                                              
Could not create surfaceless EGL display: EGL_BAD_ALLOC. Aborting...
zsh: IOT instruction (core dumped)  ./zmk-studio_0.1.0_amd64.AppImage
```

**To Reproduce**
Try to launch the AppImage

**Expected behavior**
I would expect the AppImage to launch.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment:**
 - Manjaro with KDE(Plasma 6.1.5)/Wayland, kernel 6.11.2-4-MANJARO
 - App
 - amd64 AppImage release 0.2.0

krikun98, 2024-11-12:
Also fails in an X11 session:
```
./zmk-studio_0.1.0_amd64.AppImage
Gtk-Message: 04:53:33.012: Failed to load module "xapp-gtk3-module"
Could not create surfaceless EGL display: EGL_BAD_ALLOC. Aborting...
zsh: IOT instruction (core dumped)  ./zmk-studio_0.1.0_amd64.AppImage
```
I am using an AMD 7900XT with the open amdgpu driver.

goldenHairDafo, 2024-11-12:
I have a similar issue with the AppImage. The app starts but the windows
is blank.

```
Gtk-Message: 10:37:46.524: Failed to load module "xapp-gtk3-module"
Could not create default EGL display: EGL_BAD_PARAMETER. Aborting...

** (app:8069): WARNING **: 10:38:13.563: atk-bridge: get_device_events_reply: unknown signature
```

**Environment:**
- CachyOS (Arch)
- Kernel: 6.11.6-2-cachyos
- X11/i3
- GPU: AMD 7900XT
- App-Version: 0.2.2

kmandarin-orange, 2024-11-15:
I'm also having an issue with the AppImage window being blank. When run from the terminal I see similar output, only difference is I'm missing that first line:

```bash
Could not create default EGL display: EGL_BAD_PARAMETER. Aborting...

** (app:33041): WARNING **: 17:52:39.447: atk-bridge: get_device_events_reply: unknown signature
```

##### Environment

- Fedora Linux 41 Workstation Edition
- Kernel: 6.11.7-300.fc41.x86_64
- GNOME DE (Wayland session)
- App version: 0.2.3
- Hardware:
  - Framework Laptop 13 AMD
  - CPU: AMD Ryzen™ 7 7840U
  - GPU: AMD Radeon™ 780M

petejohanson, 2024-11-16:
Can folks give the environment variables mentioned here a try? https://github.com/getyaak/app/issues/75#issuecomment-2344749813

kmandarin-orange, 2024-11-16:
Just gave that a try, still got the same errors in terminal and same viewport behavior

krikun98, 2024-11-16:
With the variables, the behavior changes - I get new errors:
```
WEBKIT_DISABLE_COMPOSITING_MODE=1 WEBKIT_DISABLE_DMABUF_RENDERER=1 ./zmk-studio_0.1.0_amd64.AppImage
Gtk-Message: 14:44:11.568: Failed to load module "xapp-gtk3-module"
Gtk-Message: 14:44:11.799: Failed to load module "xapp-gtk3-module"
Could not create surfaceless EGL display: EGL_BAD_ALLOC. Aborting...
** (app:130420): WARNING **: 14:44:36.852: atk-bridge: get_device_events_reply: unknown signature

```
And a white screen that's just open forever (as opposed to a black window that blinked on and disappeared before):
[image]

petejohanson, 2024-11-18:
Please try the downloaded artifact from #92 and let me know if it fixes things for you as well. If so, I will get it merged and released. Thanks!

kmandarin-orange, 2024-11-19:
The window contents now render correctly! Thank you for getting that working. I haven't had time to use it extensively yet but I was able to get to the studio unlock step, so that's a good sign.

krikun98, 2024-11-19:
Doesn't seem to be working for me yet, unfortunately. Will report back once I can do a bit more testing.

krikun98, 2024-11-19:
```
#from actions for PR 92
./zmk-studio_0.2.3_amd64.AppImage                                 
Cannot get default EGL display: EGL_SUCCESS
Cannot create EGL context: invalid display (last error: EGL_SUCCESS)
zsh: segmentation fault (core dumped)  ./zmk-studio_0.2.3_amd64.AppImage
#0.1.0 release
./zmk-studio_0.1.0_amd64.AppImage                                                                       
Gtk-Message: 23:41:03.998: Failed to load module "xapp-gtk3-module"
Could not create surfaceless EGL display: EGL_BAD_ALLOC. Aborting...
zsh: IOT instruction (core dumped)  ./zmk-studio_0.1.0_amd64.AppImage
```
However, `WEBKIT_DISABLE_COMPOSITING_MODE=1 WEBKIT_DISABLE_DMABUF_RENDERER=1  ./zmk-studio_0.2.3_amd64.AppImage` actually works now. Is there a way to use those variables as defaults?

petejohanson, 2024-11-19:
> ```
> #from actions for PR 92
> ./zmk-studio_0.2.3_amd64.AppImage
> Cannot get default EGL display: EGL_SUCCESS
> Cannot create EGL context: invalid display (last error: [...]

Let me research the impact before we try forcing the values universally. Thanks for testing!

petejohanson, 2024-11-19:
> If it matters, I launched the AppImage as root, not sure if that'd affect things for you but it's worth a shot

It really shouldn't require that. Can you please test running as a normal user?

kmandarin-orange, 2024-11-20:
Yeah, it does launch but it gives me issues with "permission denied" errors when selecting a device. Forgot to mention that, my bad, that was just a separate issue I worked around but need to look into.

petejohanson, 2024-11-20:
> Yeah, it does launch but it gives me issues with "permission denied" errors when selecting a device. Forgot to mention that, my bad, that was just a separate issue I worked around but need to look into.

Does your user have correct permissions to the /dev/ttyACM# device?

kmandarin-orange, 2024-11-22:
That was the issue, good catch! Had to add my user to the `dialout` group and restart, it now launches as expected and remaps keys successfully.

petejohanson, 2024-11-22:
> ```
> #from actions for PR 92
> ./zmk-studio_0.2.3_amd64.AppImage
> Cannot get default EGL display: EGL_SUCCESS
> Cannot create EGL context: invalid display (last error: [...]

I've merged the build fix. I'm *hesitant* to try to enable those by default until:

* We've tested the impact more broadly.
* We understand scope of conditions that make this required.

For now, I'd rather add this to our (planned) Studio troubleshooting documentation.