zmkfirmware/zmk-studio #86: Javascript - tauri://localhost

joelop3, 2024-11-12:
`Failed to open the serial port: Permission denied`

On:

1. Open ZMK Studio
2. Detects the keyboard
3. Click on keyboard
4. Message error appears  

[image]

**Environment (please complete the following information):**
- OS: EndeavourOS Linux x86_64
- Kernel: 6.6.60-1-lts
- DE: Hyprland
- CPU: AMD Ryzen 7 7800X3D (16) @ 5.05
- GPU: NVIDIA GeForce RTX 4070 SUPERMemory: 6359MiB / 27713MiB
- AppImage 
- Keyboard: Aurora Sofle v2 (splitkb.com) with nice!nano

Nick-Munnich, 2024-11-12:
This is most likely an error on your end rather than on Studio's. Please make sure you have the necessary permissions to connect to a usb serial port. For some linux distros, that's the `uucp` group, I don't know the details of EOS.

bickman, 2024-11-19:
On windows 24h2, I have the same issue, I can use zmk studio online version, but local version can only discover the keyboard, can't connect.

ThHareau, 2025-07-11:
Did you manage to make it work? I'm having the same issue on Fedora