zmkfirmware/zmk-studio #93: Add support for configuring combos, macros, conditional layers, and behaviors

kmandarin-orange, 2024-11-22:
The app's current feature set is functional and does what many people need at a basic level, but knowing many people use custom behaviors like mod-morph, combos, macros, and conditional layers, it'd be ideal to make that configurable in ZMK Studio. 

I'm not sure if this isn't here yet for technical limitations, or if it's feasible but needs design, if it's the latter then I think this could be solved by adding making separate pages for managing each of these actions. Entries for "Combos," "Behaviors" etc. pages could be added to the sidebar, above the layers section. As far as each page's UX flow, I think [this tool's](https://nickcoutsos.github.io/keymap-editor/) UI is a good start for how things can be handled.

That's just some off-the-cuff suggestions on how to handle UI though, so I'm not sure what's actually doable  yet, but hopefully this feature request can get us in the direction of supporting these features.

daforster, 2026-08-29:
Hello, anything new here? This feature would really be helpful since the 'keymap-editor' approach is much too sophisticated.

danielsvane, 2026-09-01:
For the time being, while waiting on official support, I have created a fork of ZMK and Studio that supports combo and behavior editing. Ive tried sharing it around, but Im quite sure Im the only one that has tested it out, so I cant promise anything. Anyway, it works for me.

You can try to build the firmware with the config template here:
https://github.com/danielsvane/unified-zmk-config-template

Firmwares:
https://github.com/danielsvane/zmk/tree/customkeyboards
https://github.com/danielsvane/zmk-studio/tree/customkeyboards

Hosted web version:
https://studio.customkeyboards.eu/

1 minute video showing the new parts, if you dont want to flash it to see it:
https://www.youtube.com/watch?v=U8Y4HVTgF8Q