zmkfirmware/zmk-studio #103: Functions for EC Keyboard

oxonium0215, 2024-12-07:
**Is your feature request related to a problem? Please describe.**
When using the ZMK firmware on an EC keyboard, I would find it useful to be able to change the depth at which the keys respond, etc. on the ZMK Studio, as Cipulot's firmware does.

**Describe the solution you'd like**
As in Cipulot's firmware, implement the following features
 - A slider or input field to change the depth to which the keys respond
 - Button to perform calibration
 - Button to toggle rapid trigger

petejohanson, 2024-12-07:
Yeah, that's definitely on the list, but there are a fair number of items that probably take higher priority. This is *likely* to be implemented either using some generic "custom settings UI" functionality, or with a general "edit arbitrary device/entity props" framework.

oxonium0215, 2024-12-08:
Thanks for the clarification! That makes sense, and a general framework does seem like the better approach in the long run. Looking forward to seeing how it develops.