zmkfirmware/zmk-studio #113: AUR Package

trevin-j, 2024-12-16:
Hey there,

I put together a working `PKGBUILD` using Tauri’s guidelines [here](https://v2.tauri.app/distribute/aur/). It builds a package for Arch using the `.deb` release. I’d like to maintain a `zmk-studio-bin` package in the AUR to make it easier for Arch users to install.

Just wanted to check in and make sure it’s okay before I go ahead and publish it. Let me know if you have any concerns!

Thanks!

bendavis78, 2024-12-22:
Hi, I'd love to try/test an AUR package. Would you mind pushing to a github repo to share?

trevin-j, 2024-12-22:
Just pushed it [here](https://github.com/trevin-j/zmk-studio-pkgbuild). Thanks!

trevin-j, 2024-12-27:
Since it’s been a while without any response, I’m going to go ahead and publish the package to the AUR under the name `zmk-studio-bin`, as it uses the prebuilt `.deb` release. If there are any concerns, please don’t hesitate to reach out. Thanks!

closed 2024-12-27 by trevin-j