zmkfirmware/zmk-studio #124: Export configuration to .keymap file

jpringle1, 2025-01-11:
Since zmk studio is currently lacking in certain features (such as creating combos), I sometimes need to configure my keyboard with a different editor (nick coutsos online editor). However, i cant do this because there is no way to get a keymap file from the uf2 files that studio uploads to the boards.

Ideally, there would be a menu option to export the existing configuration to a keymap file. Then, i could edit that file in the online editor, compile it into a uf2, upload that to the board, then resume editing in the studio.

You might ask why i dont just ignore zmk studio and do all my editing in nicks editor if i need it that much, but the reason i want to keep using zmk is because its the only editor that has the ability to modify my keyboard on the fly without having to flash anything. Id like to keep that functionality when editing basic features, then use the unnofficial editor when making more complex changes.

shchemelevev, 2025-01-19:
It would also be nice to have an import feature.

leikoilja, 2025-02-05:
this is highly important feature and is blocking adoption of zmk-studio in my workflow as well. For now i would need to manually duplicate/move configurations i did in zmk studio into my git controlled repo that is later using by nick's editor 😅

JeppeKlitgaard, 2025-02-14:
Adding to this, I think it would be nice if it could keep a keymap file local to the computer in sync with keys in order to allow keymaps to be easily managed in Git in addition to directly on the keyboard. That is, when a keymap has been 'flashed' to the keyboard, this change would also be reflected in `CONFIGURED_SYNC_FOLDER/KEYBOARD_NAME.keymap`

brandon-page, 2025-02-17:
Seems crazy this wasn't in the MVP, let alone not even mentioned as planned in the capabilities documentation. 

It's fine to not support all ZMK features and customization in the Studio GUI day 1... but that is exactly what makes this the most important feature. If all of the changes I make in Studio are lost whenever I need to change something in code it seems pretty useless.

lexxxel, 2025-02-17:
@brandon-page that is not correct. The keymap via zmk studio is saved in an area that is not override during flash. Basically the same as Bluetooth profiles. However, one can not easily use another MCU and copy the keyboard over or use custom settings for specific keys or tube timings for single keys.