zmkfirmware/zmk-studio #130: Disconnect from Bluetooth profile when not currently selected

sleevesluu, 2025-01-31:
When connected via Bluetooth to a phone, it disables the phone's on-screen keyboard, even when switching to another Bluetooth device or wired mode. The Disconnect Profile command doesn’t seem to do anything.

The best solution would be automatic disconnection from the BT profile when not currently selected, like most keyboards. 

Alternatively, fixing the Disconnect Profile command to allow for manual disconnection, which would useful for frequent device switching without having to wait for a reconnection. You can force the on-screen keyboard through phone settings, but I generally don't want that if I'm using a physical keyboard.

caksoylar, 2025-01-31:
Why is this a Studio issue? Please search the ZMK repo for related issues, there are at least a few like https://github.com/zmkfirmware/zmk/issues/1998 and https://github.com/zmkfirmware/zmk/pull/2081.

sleevesluu, 2025-01-31:
Sorry, I actually discussed this is the Discord with some people like a year ago and they said that Studio might have a solution. At the time it didn't, but I was hoping it would not.