zmkfirmware/zmk-studio #133: zmk-studio.bin looking for library in non standard path arch...

infodungeon, 2025-03-31:
I suppose I will have  to grep through the code an create symbolic link to the library but this should just work.

❯ yay -S zmk-studio-bin
AUR Explicit (1): zmk-studio-bin-0.3.1-1
Sync Dependency (1): webkit2gtk-2.48.0-1
:: (1/1) Downloaded PKGBUILD: zmk-studio-bin
  1 zmk-studio-bin                           (Build Files Exist)
==> Packages to cleanBuild?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> 1
:: Deleting (1/1): /home/robert/.cache/yay/zmk-studio-bin
HEAD is now at 7a2dae4 Update to 0.3.1
  1 zmk-studio-bin                           (Build Files Exist)
==> Diffs to show?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> n
==> Making package: zmk-studio-bin 0.3.1-1 (Mon 31 Mar 2025 11:21:15 AM CDT)
==> Retrieving sources...
  -> Downloading ZMK.Studio_0.3.1_amd64.deb...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 6295k  100 6295k    0     0  7642k      0 --:--:-- --:--:-- --:--:-- 28.9M
==> WARNING: Skipping verification of source file PGP signatures.
==> Validating source_x86_64 files with sha256sums...
    ZMK.Studio_0.3.1_amd64.deb ... Passed
:: (1/1) Parsing SRCINFO: zmk-studio-bin
[sudo] password for robert: 
resolving dependencies...
looking for conflicting packages...

Packages (2) libmanette-0.2.11-2  webkit2gtk-2.48.0-1

Total Download Size:    33.52 MiB
Total Installed Size:  126.55 MiB

:: Proceed with installation? [Y/n] y
:: Retrieving packages...
 webkit2gtk-2.48.0-1-x86_64                   33.5 MiB  21.0 MiB/s 00:02 [########################################] 100%
 libmanette-0.2.11-2-x86_64                   54.4 KiB  1026 KiB/s 00:00 [########################################] 100%
 Total (2/2)                                  33.5 MiB  19.0 MiB/s 00:02 [########################################] 100%
(2/2) checking keys in keyring                                           [########################################] 100%
(2/2) checking package integrity                                         [########################################] 100%
(2/2) loading package files                                              [########################################] 100%
(2/2) checking for file conflicts                                        [########################################] 100%
(2/2) checking available disk space                                      [########################################] 100%
:: Processing package changes...
(1/2) installing libmanette                                              [########################################] 100%
(2/2) installing webkit2gtk                                              [########################################] 100%
Optional dependencies for webkit2gtk
    geoclue: Geolocation support
    gst-libav: nonfree media decoding
    gst-plugins-bad: media decoding
    gst-plugins-good: media decoding [installed]
:: Running post-transaction hooks...
(1/1) Arming ConditionNeedsUpdate...
==> Making package: zmk-studio-bin 0.3.1-1 (Mon 31 Mar 2025 11:21:25 AM CDT)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Found ZMK.Studio_0.3.1_amd64.deb
==> Validating source_x86_64 files with sha256sums...
    ZMK.Studio_0.3.1_amd64.deb ... Passed
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Extracting ZMK.Studio_0.3.1_amd64.deb with bsdtar
==> Sources are ready.
==> Making package: zmk-studio-bin 0.3.1-1 (Mon 31 Mar 2025 11:21:26 AM CDT)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> WARNING: Using existing $srcdir/ tree
==> Entering fakeroot environment...
==> Starting package()...
==> Tidying install...
  -> Removing empty directories...
  -> Removing libtool files...
  -> Purging unwanted files...
  -> Removing static library files...
  -> Copying source files needed for debug symbols...
  -> Compressing man and info pages...
==> Checking for packaging issues...
==> Creating package "zmk-studio-bin"...
  -> Generating .PKGINFO file...
  -> Generating .BUILDINFO file...
  -> Adding install file...
  -> Generating .MTREE file...
  -> Compressing package...
==> Leaving fakeroot environment.
==> Finished making: zmk-studio-bin 0.3.1-1 (Mon 31 Mar 2025 11:21:28 AM CDT)
==> Cleaning up...
loading packages...
resolving dependencies...
looking for conflicting packages...

Packages (1) zmk-studio-bin-0.3.1-1

Total Installed Size:  14.82 MiB

:: Proceed with installation? [Y/n] Y
(1/1) checking keys in keyring                                           [########################################] 100%
(1/1) checking package integrity                                         [########################################] 100%
(1/1) loading package files                                              [########################################] 100%
(1/1) checking for file conflicts                                        [########################################] 100%
(1/1) checking available disk space                                      [########################################] 100%
:: Processing package changes...
(1/1) installing zmk-studio-bin                                          [########################################] 100%
:: Running post-transaction hooks...
(1/3) Arming ConditionNeedsUpdate...
(2/3) Updating icon theme caches...
(3/3) Updating the desktop file MIME type cache...

~ took 23s 
❯ zmk-studio
zmk-studio: error while loading shared libraries: libwebkit2gtk-4.1.so.0: cannot open shared object file: No such file or directory

~ 
❯ sudo pacman -S webkit2gtk
warning: webkit2gtk-2.48.0-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (1) webkit2gtk-2.48.0-1

Total Installed Size:  126.17 MiB
Net Upgrade Size:        0.00 MiB

:: Proceed with installation? [Y/n] Y
(1/1) checking keys in keyring                                           [########################################] 100%
(1/1) checking package integrity                                         [########################################] 100%
(1/1) loading package files                                              [########################################] 100%
(1/1) checking for file conflicts                                        [########################################] 100%
(1/1) checking available disk space                                      [########################################] 100%
:: Processing package changes...
(1/1) reinstalling webkit2gtk                                            [########################################] 100%
:: Running post-transaction hooks...
(1/1) Arming ConditionNeedsUpdate...

~ took 2s 
❯ zmk-studio
zmk-studio: error while loading shared libraries: libwebkit2gtk-4.1.so.0: cannot open shared object file: No such file or directory

VSladek, 2025-04-08:
had the same problem, it looks like the `webkit2gtk-4.1` is a missing dependency in aur

MrDwarf7, 2025-07-09:
I have notified the AUR maintained via a comment, hopefully it gets updated soon.

petejohanson, 2025-07-09:
Can we close this then? Any outstanding concerns here?

MrDwarf7, 2025-07-11:
@petejohanson Meant to get back to this later in the evening yesterday- 

I've not yet had a chance to test in a clean-state environment, but it was also the only action I personally had to take to have the package behave as intended. 
That being said - with the update (Which after seeing the response, I pulled from the AUR) I can see the dep. that's required is listed and it built just fine.

tl;dr: dep. correctly listed in AUR & works on my machine.

ps: Pete - I see you're the primary contributor/owner for the repo; I'd like to contribute on the Rust side of the codebase. Is there a roadmap/project direction so I can better understand what can be implemented (or moved - in performance critical/safety context) to the Rust backend and assist with it?